GaudiLabs/OpenThereminV4 #1: More flexibility on USB/Serial interface design could be nice

MrDham, 2021-03-25:
Hello Urs, 

I had a look at this new theremin schematic. It is a very nice idea to merge arduino and schield board in a single one. 

I hope that I don't bother you in a too much early or too much late stage of the development, I have just one comment: 

As CH340G can't be re-programmed into a MIDI USB port, in order to allow implementation of some additionnal material for a MIDI interface (MIDI DIN connector, ... ), an additionnal free solder pad on TXD could be very nice to have. 

Regards

MrDham, 2021-09-10:
It seem that this issue is fixed, isn't it ?

In most recent design, I can see that a new pad is connected to RXD of CH340G wich is equivalent to TXD of ATMEGA 328P -> correct :+1:

GaudiLabs, 2021-12-09:
Yes, we can now do MIDI over serial and real MIDI connector to TX pad.