GaudiLabs/OpenThereminV4 #4: PCB file is inconsistent with GBR file.

ws786873, 2022-03-06:
PCB file is inconsistent with GBR file. I want to use PCB file to generate SMT positioning, but it seems that PCB file is a semi-finished product. 0603 is a little difficult for me.

GaudiLabs, 2022-03-06:
Hi
Before just accusing me of publishing unfinished files, please tell me who you are and how you can contribute in joining this project. Then please be  specific enough for me to see where you ran into problems. If you just want a working theremin you can buy it here:

gaudishop.ch

Best,
Urs

🧪 GaudiLabs LLC
Urs Gaudenz
Email: ***@***.***
www.gaudi.ch

ws786873, 2022-03-06:
Sorry, I just want to make one. I can't seem to contribute anything to the project. Sorry again.

GaudiLabs, 2022-03-06:
The files are designed on the open design software KiCad. You can export Gerber files from thete. If you see inconsistencies with the repo files,  please regenerate the gerber files and fix that
Urs

🧪 GaudiLabs LLC
Urs Gaudenz
Email: ***@***.***
www.gaudi.ch

closed 2022-03-07 by GaudiLabs