GaudiLabs/OpenThereminV4 #11: Problem with a capacitor in the BOM

emilien-g, 2023-09-13:
Hi, 
I think there's maybe a mistake in the BOM
On line 31, for "C15,C16,C17,C18,C19,C31,C34", it's written both "22pF 16 V X7R 5%" and "MUST BE C0G (NP0)"
Thanks a lot for your work !

krokosik, 2024-08-12:
@GaudiLabs @emilien-g I want to order the capacitor, but I'm not sure which variant is right. Which is it?

krokosik, 2024-08-12:
From what I can see in the schematic, these are used for decoupling near the the button, while the others form Pierce Oscillators together with the crystals. I am no expert in audio, but from what I now the C0G are only needed in the LC oscillator, and for the aforementioned applications, the cheaper X7R is sufficient.