GaudiLabs/OpenThereminV4 #13: No output through usb serial

metehankutlu, 2024-01-03:
I'm trying to connect my v4 through usb serial but I'm not getting any data from the theremin.

[image]

I am sure about "/dev/ttyUSB0" because it's the only available port. It's also doesn't show anything when I use ttymidi. What am I doing wrong?

MrDham, 2024-01-03:
Hello, 

Out of the box, Open Theremin V4 doesn't come with MIDI. 
The serial port is there to monitor calibration and as such is just a serial port. 

If you downloaded the MIDI software into your open theremin V4 and a seeking for some support, then you should rather ask this question in my repo https://github.com/MrDham/OpenTheremin_V4_with_MIDI.

metehankutlu, 2024-01-04:
Oh it says "OpenTheremin **with MIDI software** is sending MIDI..." I haven't read that correctly. I thought there was something wrong with my theremin, thanks!

closed 2024-01-04 by metehankutlu