GaudiLabs/OpenThereminV4 #15: tonal distribution

twentyfortysix, 2024-09-04:
Hi, sorry for the question in the issues.. but might be something useful. 

Is it possible to define the tonal distribution across the pitch range?

scenario:
The lower tones are bigger therefore has wider hand movement range while the high tones are getting closer and closer and are harder to play as the distance between them is getting smaller.
What would be useful would be to have a chance to alter the values so that the lower tones are closer than are by default and the higher tones are further form one another than are by default.
It might be on a software side as function that would adjust the incoming pitch value by a factor value, the value distribution

python example
```
import numpy as np

def exponential_distribution(arr, factor=1.0):
    """
    Adjust the distribution of numbers in an array to follow an exponential curve.
    
    Parameters:
    arr (list of float): The input array of numbers.
    factor (float): The factor controlling the steepness of the exponential curve.
                    Values > 1.0 make the curve steeper, values < 1.0 make it shallower.
    
    Returns:
    list of float: The adjusted array of numbers following an exponential distribution.
    """
    if len(arr) < 2:
        return arr  # No adjustment needed for arrays with less than 2 elements
    
    # Normalize the input array to the range [0, 1]
    normalized_arr = np.linspace(0, 1, len(arr))
    
    # Apply the exponential transformation
    exp_arr = (np.exp(normalized_arr * factor) - 1) / (np.exp(factor) - 1)
    
    # Scale the exponential array back to the original range
    min_value = arr[0]
    max_value = arr[-1]
    adjusted_arr = min_value + exp_arr * (max_value - min_value)
    
    return adjusted_arr.tolist()

# Example usage
original_array = [0, 1, 2, 3, 4, 5, 6, 7, 8, 9]
factor = 2.0  # Increase the steepness of the exponential curve
adjusted_array = exponential_distribution(original_array, factor)
print("Original array:", original_array)
print("Adjusted array:", adjusted_array)
```

result
```
Original array: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9]
Adjusted array: [0.0, 0.3505431453935727, 0.7883185560546282, 1.3350338825354344, 2.017798699676874, 2.870468769358103, 3.935324821511089, 5.265169097891943, 6.9259436183985335, 9.0]

```

see the value distribution changes across the range, and that what is needed.
The Open theremin would have either a function with a factor value we can change as a number or the better a knob that would allow us to alter the distribution.

As that is the killer feature the PRO Moog theremins used to have, which makes them way more playable as the high notes thanks to the altered pitch value were easier to be played.

closed 2024-09-05 by GaudiLabs

twentyfortysix, 2024-09-05:
Completed?
how is that completed? Would be nice to know how it is implemented if that is implemented already.

GaudiLabs, 2024-09-05:
Hi

Thanks for sharing your thoughts for a tonal distribution feature. The micro-controller on the current version is quite loaded already and I don't think it could do the necessary calculations fast enough. You could try of course :-). However I am collecting ideas for a next version with a more speedy processor and will for sure keep this in mind.

Best,
Urs

twentyfortysix, 2024-09-05:
Hi Urs
Just an idea, kinda brute force. What about using the mapping array, similar to wavetable. 
Actual value would simply by mapped to whatever is in the mapping array. That way there will be no calculation.. 

Although it seems that one value processing should not be that expensive.

I have checked the code a bit and it looks like you have some pitch factors already
https://github.com/GaudiLabs/OpenThereminV4/blob/main/Software/Open_Theremin_V4/application.h 
the HZ_ADDVAL_FACTOR looks like it would do the trick actually (application.cpp).

```
void Application::hzToAddVal(float hz)
{
  setWavetableSampleAdvance((uint16_t)(hz * HZ_ADDVAL_FACTOR));
}
```

here is a modified version with a exponential factor in the play

```
// Define the exponential factor
const float EXPONENTIAL_FACTOR = 2.0f; // Example value, adjust as needed

void Application::hzToAddVal(float hz)
{
  // Apply exponential transformation
  float transformedHz = pow(hz, EXPONENTIAL_FACTOR);
  
  // Calculate the sample advance value using the transformed frequency
  setWavetableSampleAdvance((uint16_t)(transformedHz * HZ_ADDVAL_FACTOR));
}
```
see it is easy ;) One variable and here we go, a flexible tonal distribution adjustment I have never been able to adjust on the Moog :)

Well Once the unit arrives I'll give it a go and will fiddle around with the code try what I can do about it.

Btw. I really like the Open theremin Idea and I feel like it is the actual utmost Theremin out there with lot of potential with all the flexibility thanks to the open code.